olistrik/goskew #4: Script introduces errors into GCode - unusable output

eparadis, 2022-08-30:
Hi, first thanks for writing this. I love the idea and it is easy to use.

Unfortunately this script is introducing errors into the Gcode it processes.

Here is a screen shot of the last layer of a print as shown in Octoprint's Gcode viewer.
[image]

Here is the same gcode after processing through `goskew`.
[image]

You can clearly see the "hitches" in the second screenshot that are not supposed to be there. These are real - they do show up in the final print quite noticeably.

I'm not great at reading Gcode but I think you can see one of the hitches here:
```
;TYPE:External perimeter
G1 F1500
G1 X85.378 Y93.315 E.02112
G1 X85.074 Y92.903 E.02113
G1 X84.814 Y92.463 E.02112
G1 X84.594 Y92 E.02113
G1 X85 Y91.517 E.02112    // <--- note this line!
G1 X84.306 Y91.019 E.02112
G1 X84.235 Y90.512 E.02114
```

That should be the top left part of a circle, so you would expect the X coordinate to decrease smoothly, but that one lines has a "X85" for some unknown reason.

I think this is the same section in an unprocessed file:

```
;TYPE:External perimeter
G1 F1500
G1 X85.961 Y93.315 E.02112
G1 X85.655 Y92.903 E.02113
G1 X85.392 Y92.463 E.02112
G1 X85.172 Y92 E.02113
G1 X85 Y91.517 E.02112     // <--- original this line!
G1 X84.875 Y91.019 E.02112
G1 X84.8 Y90.512 E.02114
```

Perhaps there is an issue with integers being used instead of floating point? I don't know how `go` works with numerics.

I suspect this is an easy fix - perhaps forcing a type on whatever reads the input.

eparadis, 2022-08-30:
I think the error is in the regex.

eparadis, 2022-08-30:
I think the error is in the regex.

[This](https://github.com/Kranex/goskew/blob/47285f8ecbd1643ba7d5c4d3cadf2b42848d4cdc/goskew.go#L57) won't match an integer because it assumes a decimal point.
`\[xX](-?\d*\.\d*)`

Quick example cases [here](https://regex101.com/r/qL0U7D/1)

eparadis, 2022-08-30:
this PR fixes my situation https://github.com/Kranex/goskew/pull/5

Kranex, 2022-10-17:
@eparadis sorry I wasn't notified of this issue or the PR for some reason. I just had the most horrific print as a result of this, I'm amazed that I didn't notice this sooner as I skew literally everything. I'll merge in your PR now.

Kranex, 2022-10-17:
merged and released to v1.3.1. Thank you for the fix!

closed 2022-10-17 by Kranex