olistrik/goskew #8: Gcode error

kiddocteur, 2023-12-29:
Hello, thank you for your work!
I have got now quality issue, we can see that parasitic movements are created.
Do you have any idea why? 
Here's the code I put in prusaslicer : /Applications/goskew-darwin-amd64/darwin-amd64/goskew err 0.00316106 0.00236537 --yz=-0.00221065;

Best regards
[image]

Dimitri

olistrik, 2023-12-29:
@kiddocteur wow, that's a first. I've seen layer shift errors in the past when prusa decided to drop decimals on whole numbers, but this looks like the scale has been changed by a factor of 100 or so.

Given that all the affected lines are arcs or circles, my first guess is those might be G2 or G3 instructions. The math for those was much harder, and I've never seen a slicer produce them, so I figured it'd be safe to simply ignore them, maybe I should have had the script error instead.

Can you slice it without skewing it and check if the gcode contains any G2 or G3's?

If it doesn't contain any, would you be able to share a copy of the guide both before and after skewing?

olistrik, 2023-12-29:
Alternatively I've noticed you've got a negative yz skew parameter, can you try skewing without it? A lot of those arcs seem to be flipped from their origin so it could also be related to that.

kiddocteur, 2023-12-29:
Hello ! 
I do indeed have G2 and G3 in the Gcode. I use prusaslicer 2.7.1 with MK4
I tried : with and without input shaper, with and without negative value, I have every time this results 
Thank you for your answer
Dimitri

olistrik, 2023-12-29:
Then I'm guessing it's related to the G2/3 codes. For now G2/3 generation can be disabled in `Print Settings->Advanced->Slicing->Arc fitting`.

If that fixes it for you I'll work on adding an arc breaker option to go-skew, and maybe include a warning. I'm no mathematician, but given how the G2/3 commands describe _perfect_ arcs, I don't think skewing them will be possible without breaking them into multiple smaller arcs, and even then i don't think skewing across all planes simultaneously would be possible as it linearly interpolates the z position. 

Perhaps a tool like arc welder could be used to re-arc the gcode after skewing if necessary.

kiddocteur, 2023-12-30:
it's perfect! 
I've deactivated arc fitting, there are no more errors, and I've been able to integrate ArcWelder, which has added G2/G3 without creating any anomalies.
Thanks again for the script and the help!

olistrik, 2024-01-22:
@kiddocteur No problem!

I've also noticed that prusa has started using binary gcode which is also throwing a spanner in the works.

olistrik, 2024-05-23:
@kiddocteur I've been thinking about this. My intent with GoSkew is that it doesn't alter the gcode, it only skews the coordinates of each instruction. I may be wrong, but I don't think it's mathematically possible to skew the arcs without breaking them in to sub arcs (or just into lines like they'd be without G2/3). But I don't want GoSkew to be adding or removing instructions so I don't think that's an option.

I think the best I can do is throw an error if a G2/3 is encountered so that you don't unknowingly skew a file with G2/3's. I'll test to see if prusaslicer will abort if a post-processing script errors.

olistrik, 2024-05-25:
prusaslicer will abort, it'll also display enough of an error log that I can provide a warning message:

[image]

olistrik, 2024-07-16:
I've discovered G5 which describes bezier curves. I'm not sure if it's supported by all printers, but not only are bezier curves easy to skew, it should be pretty straightforward to convert any G2/3 movements into G5 movements.

GCheung55, 2024-07-17:
> it's perfect! I've deactivated arc fitting, there are no more errors, and I've been able to integrate ArcWelder, which has added G2/G3 without creating any anomalies. Thanks again for the script and the help!

How do you add ArcWelder? Disable arc fitting, then add ArcWelder in the post-processing scripts? I’m assuming after goSkew?

olistrik, 2024-07-17:
@GCheung55 I haven't tried it myself, but yes, after goSkew would be correct. You can either disable arc fitting, or you could run arc straightener first, though that would be a little wasteful on resources:

```sh
# C:\{PATH_TO_ARCWELDER_LIB}\ArcStraightener.exe; # or just disable arc fitting.
C:\{PATH_TO_GOSKEW}\goskew err --xy=[XYTAN] --xz=[XZTAN] --yz=[YZTAN];
C:\{PATH_TO_ARCHWELDER_LIB}\ArcWelder.exe;
```

I should warn you though, arc welder only supports G2/3 movements, and after skewing it's very unlikely that any arc can be easy described with a perfect arc. For example, if you have a perfect half circle, rather than describing that as a single G2/3 movement, it'll need to be broken several. the more sections required, the less benefit you get. Depending on the settings and how much skew we're talking, there's also the possibility that arcwelder will decide on a "good enough" arc and it'll effect your tolerances.

There's no harm in trying though.

I ran a quick test, if I take the perfect circle:

```gcode
G2 I20 J20
```

Piping it through ArcStreightener and ArcWelder without skew results in:

```gcode
G2 X0.722 Y-0.697 I20.003 J19.998
G1
```

skewing in between with an `xytan: 0.0170843`:

```gcode
G2 X-5.936 Y32.482 I19.874 J20.415
G2 X26.756 Y47.254 I25.120 J-12.032
G2 X47.103 Y12.679 I-7.616 J-27.757
G2 X14.029 Y-7.618 I-26.940 J6.803
G2 X0.734 Y-0.697 I6.224 J28.186
G1
```

This reduces the straightened gcode from 176 G1 movements to 5 G2 movements. Can't comment on the loss of accuracy though.