olistrik/goskew #13: Skew it cube model

BackwardEcho, 2025-04-27:
More of a question / confirmation about how to use the "Skube" model (since I am not sure where else to put my question): Am I to measure against the labeled edges like this? Does the model of the cube actually matter as long as I am taking proper measurements of the same diagonal? I am thinking of redesigning the cube for myself so I can measure against flat faces instead of hard edges or seeing if Vector 3D's "Calilantern" will work.

[image]

olistrik, 2025-04-27:
@BackwardEcho Perfect place for questions! I guess technically the [issue board for skew-it](https://github.com/olistrik/skew-it/issues) would be more correct, but that's much harder to find.

You are correct, it's the labeled hard edges, unless you measure the faces (X, Y, and Z) or the chamfered corners (G,H,I, and J). 

At the time, I didn't think about it to much. I chose hard edges because I wasn't sure if parallel edges would remain parallel after being skewed, but that should be the case. If it's not you've got bigger problems that goskew can't fix. 

I also didn't know how poor surface quality or shrinkage would effect things. But assuming uniformity in both, then skew-it shouldn't care. 

Skew-it uses the ratio between orthogonal measurements for it's calculations, so if you did flatten the edges it shouldn't matter.

I can add a skube with chamfered edges to skew-it tomorrow as I should have the original model somewhere.

BackwardEcho, 2025-04-27:
@olistrik Oh cool, thank you! You don't need to spend too much time on finding / altering the skube as I may just use the previously mentioned [Vector 3D "Calilantern"](https://vector3d.shop/products/calilantern-calibration) as its measurements 9, 10, 11, 12, 13, and 14 seem to correspond to your own A, B, C, D, E, and F measurements. It's why I asked if the model mattered or if I could just get the proper measurements.

Aside from that, what sort of values does the Skew-It calculator put out? Are they only for the goskew program or are they values that Klipper, Marlin, or RepRap could use? While I could use the lantern for measurements 9 to 14 for your A to F and use your calculator, it does have it's own calculator to spit out values for those specific firmware options but it would require measurements 1 to 8 to properly do so. However my printer runs on Chitu (I don't want to fiddle with firmware modding on this machine yet) so it would be pointless for me to use that function unless your program accepts the values for one of those firmwares.

olistrik, 2025-04-28:
@BackwardEcho it's no problem! I probably won't get around to updating it on the skew-it webpage today but [I've added the altered skube to the skew-it repo](https://github.com/olistrik/skew-it/blob/main/public/skube_v2.stl).

The measurements from the calilantern should be totally fine too.

olistrik, 2025-04-28:
As for the values themselves, they are computed as:

$$
\epsilon = \tan \left( 2 \cdot \arctan \left(\frac{a}{b} \right) - \frac{\pi}{2} \right)
$$

For example, in the XY plane, it describes the amount of error in the X axis, for each mm traveled in the Y axis. Go skew uses this to calculate the corrected X position like so:

$$
x = x - y \cdot \epsilon_{xy}
$$

Klipper skew correction doesn't use these, [instead it uses the measurements directly](https://github.com/MechanizedMedic/gskewer). It also seems to require the width even though mathematically it's not required. This implies that their math depends on the expected dimensions of model, so I would be hesitant to use any calibration model other than the one they provide.

[Marlin uses the M852 gcode](https://marlinfw.org/docs/gcode/M852.html), I'm having trouble finding documentation on how these should be calculated, but based on [this line](https://github.com/MarlinFirmware/Marlin/blob/75d60b77bac5300e38725c0aae6f788b70ef085b/Marlin/src/module/planner.h#L753) in the firmware they're the same as goskew. However [this line](https://github.com/MarlinFirmware/Marlin/blob/75d60b77bac5300e38725c0aae6f788b70ef085b/Marlin/src/module/planner.h#L763) would suggest that they are the exact opposite, so you would need to multiply by `-1`. I don't know if marlin "skews" or "unskews" to compensate.

[RepRap (and Duet3D) uses M556](https://reprap.org/wiki/G-code#M556:_Axis_compensation). If you use an S value of 1, then it should be the same as values from skew-it.

BackwardEcho, 2025-04-29:
[I was linked to this line in Marlin](https://github.com/MarlinFirmware/Marlin/blob/7503ac3c67b1521f8d2c822f51b46c97a884a21e/Marlin/Configuration.h#L2162) when I asked for a skew compensation feature in Cura. Unfortunately, Cura is lame and don't like to add cool features to their slicers UNLESS their specific printers benefit the most from such a feature. Does this line / section help you figure out how Marlin calculates it?

I am going to bite the bullet and purchase a copy of V3D's calilantern, especially since it is highly likely more of the packaged calculator functions might be applicable to goskew.

olistrik, 2025-04-30:
In my experience skew correction is not a feature of any slicer, and you don't really want it to be either. Ideally it should be configurable on the printer, but not all printers allow you to flash custom firmware which is why I made GoSkew. 

Cura and most other slicers support adding [additional gcode instructions to start / end](https://github.com/Ultimaker/Cura/wiki/Start-End-G%E2%80%90Code) of each print file, which is one place that you can add the `M852 ...` instruction to configure the skew compensation. Though the `M852` instruction will only work if `SKEW_CORRECTION_GCODE` was defined when the printers firmware was compiled, otherwise it will simply be ignored.

The "Marlin G-Code Adjustment" section of your screenshot has shows that there is another way to do this independently of the slicer that I hadn't considered; If the `M852 ...` code is enabled in firmware, then you can send it to the printer over the serial console, then follow it up with `M500` which will save the new skew compensation values so that all future prints will use them.

The RepRap firmware seems to be compiled with `M556` enabled by default, so it only needs to be included in the `config.g` file which I'm guessing is on the printer, or the flash medium that it's printing from (I don't have a RepRap printer so I don't know).

I'm intrigued as to what you automated, was it skew-it or GoSkew?

BackwardEcho, 2025-04-30:
While I may not be versed in reasons why a slicer should or shouldn't have some form of skew correction, they can still do so in a very basic way: which would be to adjust the model itself prior to slicing. While I was testing the script I put some bs skew factors and got G0 coordinates of -1800. Obviously this is not something a printer should ever do, but it shows that a post process cannot account for machine limits. Should the slicer itself be the one to compensate skew, then the gcode generation could be confined to the printer's max build space with some checks to ensure there are no moves beyond.

I don't think there would be anyway to automate the use of the Skew-it site. The script automates the use of the GoSkew program: searches a defined folder for any .gcode files without a "Skewed" suffix, if it finds one it uses the command line and starts GoSkew to process the file, afterwards it renames the code file with "_Skewed" at the end. If it finds a file with "Skewed" in the name then the file is skipped. It can do multiple gcode files with one activation of the script.

olistrik, 2025-04-30:
The main reason I say that the slicer shouldn't do it is because any skew presented by a printer is a property of that specific printer. Ideally rather than having to mess around with machine codes and custom firmware, it should just be an entry in the settings menu on the machine itself. My reasoning for this is that skewing individual files, for individual machines becomes quite frustrating if you have more than one of the same printer as each one requires it's own personal slicing configuration, and it's own personal file to print even if they're all printing the exact same model.

> Should the slicer itself be the one to compensate skew, then the gcode generation could be confined to the printer's max build space.

The slicer would still need to skew the model beyond the limits of the machine to compensate. It's a known problem with skewing and printers will usually just ignore gcode that goes beyond their limits, or they will stop when they reach a limit. At least they _should_. The concern with skew compensation isn't usually reaching your printers limits, but colliding with things like the alignment posts on the prusa mini's print bed. Personally I would argue that It's more likely to collide with those while printing without correction, because clearly the print head is not where it actually intends it to be. 

Personally I consider it something important to consider, but ultimately an unlikely situation to worry about. If your printer is so poorly aligned that you're more 1-2mm out over the length of your print bed then would be better fixed with shims. Skew calibration is for fixing that last 0.1mm of skew when you're printing functional parts or large objects that require that level accuracy. 

GoSkew was intended to be used as a postprocessing script, in prusaslicer at least you can configure it to run GoSkew automatically for every model sliced. While Cura _does_ support postprocessing scripts, the only examples I can find are in Python, and I can't find any documentation on them, so your script is probably the best way! 

I had considered making GoSkew in a way that dragging a file onto the executable would spit out a skewed version, but ultimately decided that if anyone wanted something fancy, making a wrapper script like what you've created would be better than a "one size fits all" solution baked into GoSkew.

BackwardEcho, 2025-04-30:
I actually was asked about some prints that should have been the same gcode to the same model of printer with the same brand of filament, but one came up underextruded and I think the other had a really squished first layer. I told them to treat the printers as their own specific machine since even though a lot of it is the same, something is always different enough to warrant the extra work. I agree it should be done on the firmware side, but even if someone's printer technically supports it, there could be reasons why it is not enabled or prevented. For example I have a delta printer with modded Marlin firmware. Marlin supports Skew Compensation as we have already discussed, however the firmware is on the stock controller board that came with the printer and there is limited space in storage so many features of Marlin had to be left out to fit what is there. I recently asked the firmware maintainer how much room is left on the board for Skew Compensation but I haven't heard back yet. I think having it done via the slicer means more long term support for an alternative method of compensation with the added benefit of checks in the gcode that a post process cannot do. 

I just it would be a good quality of life improvement, to little to no drawbacks, if slicers had the option.

This script should cover all other slicers then as it is just about the program file path and chosen folder. I just have a shortcut that will forever live in my printer's SD card since I already check that often when I'm slicing something to make sure I didn't forget to save it as "Chitu Gcode" in Cura (so I get that image preview) and I'm not sure what a .bat file will do to the Chitu Firmware my printer has.

Saved it as a .txt file since this is my first time really dealing with a script .bat file extension, so feels less sketchy to me to share it this way, and I guess it kind of forces the user to look at it first before they use it.

[Skewing Script.txt](https://github.com/user-attachments/files/19980586/Skewing.Script.txt)

BackwardEcho, 2025-05-01:
Well, I am really glad your program takes Marlin Skew Factors without adjustment. I was able to obtain a score of ~964 out of 1000 on the calilantern calculator. I highly recommend it as it also suggests material shrinkage values for the slicer to fix the size of the printed model to get it ever closer to perfection.

I guess with this I'll close this "issue" out so it isn't just left open after my questions / concerns have been answered. Thank you.

closed 2025-05-01 by BackwardEcho